adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #28: ESP8266 support

lunanigra, 2015-10-23:
Is the Adafruit-PCD8544-Nokia-5110-LCD lib already fully ESP8266 compatible?

Following lines for example have caused a compiler issue (Adafruit_PCD8544 193ff.)
```
clkport     = portOutputRegister(digitalPinToPort(_sclk));
clkpinmask  = digitalPinToBitMask(_sclk);
mosiport    = portOutputRegister(digitalPinToPort(_din));
mosipinmask = digitalPinToBitMask(_din);
```

bbx10, 2015-10-24:
You are welcome to try this fork modified for the ESP8266.

https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library

lunanigra, 2015-10-25:
@bbx10 : Thanks your fork works!

As done with the GFX lib it would be cool if the Adafruit PCD8544 lib could be modified as well.

mcauser, 2016-03-01:
@bbx10 works for me too. Thanks!

canislupus11, 2016-03-27:
Hi!
I have Arduino v1.6.8 and fork for ESP of Adafruit PDC8544 library (https://github.com/bbx10/Adafruit-PCD85 ... ee/esp8266)

When i compile and upload example code "pcdtest" i get

` ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
`

WereCatf, 2016-03-27:
@canislupus11 Did you change the pins in the example?

canislupus11, 2016-03-27:
No. But I should?

WereCatf, 2016-03-27:
@canislupus11 Obviously.

bbx10, 2016-03-27:
@canislupus11 I replied to your message on esp8266.com.

moxy-bc, 2016-04-03:
When I run your test application on ESP8266 I get:

moxy-bc, 2016-04-03:
When I run you pcdtest sketch it produces a fatal error:  

Arduino\libraries\Adafruit_PCD8544\Adafruit_PCD8544.cpp:20:26: fatal error: avr/pgmspace.h: No such file or directory

I changed the pins to match my hardware.  Do I have to define something else to make this work on ESP8266?

WereCatf, 2016-04-03:
@moxy-bc: did you grab the esp8266 - branch of bbx10's version of the library, or did you just grab the master - branch? (git clone -b esp8266 https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library.git)

moeins, 2016-09-17:
The Library https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library.git  did not work!
1. fatal error: avr/pgmspace.h: No such file or directory
when I fixed it with:
#if defined ( ESP8266 )
  #include <pgmspace.h>
#else
  #include <avr/pgmspace.h>
#endif

It comes with this error:
error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     clkport     = portOutputRegister(digitalPinToPort(_sclk));
                 ^
error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     mosiport    = portOutputRegister(digitalPinToPort(_din));

WereCatf, 2016-09-17:
@moeins You didn't grab the esp8266 - branch, you grabbed the master - branch.

moeins, 2016-09-17:
@WereCatf 
Where can I find the esp8266 - branch?!?

witnessmenow, 2017-01-07:
Hi All,

I'm using @WereCatf fork, using the correct esp8266 branch and the example works fine when I have it wired up.

When I tried to use it another sketch it was crashing on display.begin(), after a lot of commenting out I narrowed it down to line Wifi.begin(ssid, pass) in my own sketch. If this line is in my sketch (either before or after calling display.begin() the crash happened.

`Exception (9):
epc1=0x40203184 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe838d depc=0x00000000

ctx: cont 
sp: 3fff1920 end: 3fff1b60 offset: 01a0

>>>stack>>>
3fff1ac0:  00000004 00000028 3fff094c 40202f97  
3fff1ad0:  3fff094c 3ffe838d 000001f8 402031ec  
3fff1ae0:  00000004 00000028 3fff0668 000001f8  
3fff1af0:  3ffe838d 00000000 3fff0668 40204d90  
3fff1b00:  402010ae 000001f4 3fff0668 3fff0b2c  
3fff1b10:  00000004 00000028 3fff0668 40204f26  
3fff1b20:  3fffdad0 00000000 3fff0a70 40201f2e  
3fff1b30:  feefeffe feefeffe feefeffe feefeffe  
3fff1b40:  3fffdad0 00000000 3fff0b24 402057fc  
3fff1b50:  feefeffe feefeffe 3fff0b40 40100718  
<<<stack<<<
`

Has anyone else ran into this? Esp8266 will try to connect to your last used credentials so removing the Wifi.begin is an option on a board that I have already programmed my credentials on but to use display on a new esp8266 I'm going to first have to write another sketch to it first (which is not the end of the world or anything!)

You can recreate by adding the following to the example:
```

#include <ESP8266WiFi.h> //to the headers

//including the above seemed to lose reference to min, so here is what i presume the implementation is, not exactly important anyways.
int16_t min(int x, int y){
  if(x < y)
  {
    return x;
  }

  return y;
}

void setup()   {
  Serial.begin(9600);

  //delay(1000);
  WiFi.begin("ssid", "pass");
....
```
Thanks for the work on the library, other than this it works great!

I'm using an official wemos mini board.

bravenius, 2017-06-14:
Nothing worked for me, seems Adafuit library and ESP8266 don't like each other... no workaround let program compile. Finally I switched to other 5110 library (not the best as doesn't support little fonts) but got it compiled and worked. Normally Adafruit libraries are tricky to implement or maybe hardware specific.

witnessmenow, 2017-06-14:
@werecatf branch works for me on the esp8266 with the latest release version of esp8266 core

bravenius, 2017-06-14:
OK, maybe need to update core library. using NodeMCU 1.0, give it a try again today. Thanks @WereCatf

innesm, 2018-02-09:
The link posted above by bbx10 is the wrong link - it isn't modified for the esp8266. Here is a link to his/her esp8266-specific branch (have compiled it, but haven't tried running it):

[esp8266 branch](https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library/tree/esp8266)

infoyobots, 2018-03-06:
@innesm 
It works!!!!! 
Thanx a lot.

cihadkck, 2019-07-24:
when i run pcdtest example, i am having the NodeMCU 0.9 (ESP-12 Module) compiling error. I've already dowloaded it from board manager. I've used ESP8266WiFi.h library. Where is the problem?

`

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_PCD8544.h>

 Adafruit_PCD8544 display = Adafruit_PCD8544(D0, D1, D2);

#define NUMFLAKES 10
#define XPOS 0
#define YPOS 1
#define DELTAY 2

#define LOGO16_GLCD_HEIGHT 16
#define LOGO16_GLCD_WIDTH  16

static const unsigned char PROGMEM logo16_glcd_bmp[] =
{ B00000000, B11000000,
  B00000001, B11000000,
  B00000001, B11000000,
  B00000011, B11100000,
  B11110011, B11100000,
  B11111110, B11111000,
  B01111110, B11111111,
  B00110011, B10011111,
  B00011111, B11111100,
  B00001101, B01110000,
  B00011011, B10100000,
  B00111111, B11100000,
  B00111111, B11110000,
  B01111100, B11110000,
  B01110000, B01110000,
  B00000000, B00110000 };

void setup()   {
  Serial.begin(9600);

  display.begin();
  
  display.setContrast(50);

  display.display(); // show splashscreen
  delay(2000);
  display.clearDisplay();   // clears the screen and buffer

 
  display.drawPixel(10, 10, BLACK);
  display.display();
  delay(2000);
  display.clearDisplay();

  testdrawline();
  display.display();
  delay(2000);
  display.clearDisplay();

 
  testdrawrect();
  display.display();
  delay(2000);
  display.clearDisplay();

 
  testfillrect();
  display.display();
  delay(2000);
  display.clearDisplay();

[... 216 lines cut ...]
  }
  delay(250);
  
  display.clearDisplay();
  for (int16_t i=display.width()-1; i>=0; i-=4) {
    display.drawLine(display.width()-1, display.height()-1, i, 0, BLACK);
    display.display();
  }
  for (int16_t i=display.height()-1; i>=0; i-=4) {
    display.drawLine(display.width()-1, display.height()-1, 0, i, BLACK);
    display.display();
  }
  delay(250);

  display.clearDisplay();
  for (int16_t i=0; i<display.height(); i+=4) {
    display.drawLine(display.width()-1, 0, 0, i, BLACK);
    display.display();
  }
  for (int16_t i=0; i<display.width(); i+=4) {
    display.drawLine(display.width()-1, 0, i, display.height()-1, BLACK); 
    display.display();
  }
  delay(250);
}`

drak7, 2019-07-29:
ESP8266 support added in #47. 

@cihadkck What specific error are you getting?

ta4ukoma, 2019-07-31:
> Is the Adafruit-PCD8544-Nokia-5110-LCD lib already fully ESP8266 compatible?
> 
> Following lines for example have caused a compiler issue (Adafruit_PCD8544.cpp 193ff.)
> 
> ```
> clkport     = portOutputRegister(digitalPinToPort(_sclk));
> clkpinmask  = digitalPinToBitMask(_sclk);
> mosiport    = portOutputRegister(digitalPinToPort(_din));
> mosipinmask = digitalPinToBitMask(_din);
> ```

Hello! I have a same problems. Compiler said:
.pio/libdeps/nodemcuv2/Adafruit-PCD8544-Nokia-5110-LCD-library/Adafruit_PCD8544.cpp: In member function 'void Adafruit_PCD8544::begin(uint8_t, uint8_t)':
.pio/libdeps/nodemcuv2/Adafruit-PCD8544-Nokia-5110-LCD-library/Adafruit_PCD8544.cpp:187:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     clkport     = portOutputRegister(digitalPinToPort(_sclk));
                 ^
.pio/libdeps/nodemcuv2/Adafruit-PCD8544-Nokia-5110-LCD-library/Adafruit_PCD8544.cpp:189:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     mosiport    = portOutputRegister(digitalPinToPort(_din));
                 ^
*** [.pio/build/nodemcuv2/liba25/Adafruit-PCD8544-Nokia-5110-LCD-library/Adafruit_PCD8544.cpp.o] Error 1

Can anyone help with that? I'm trying to compile from VScode with platformio

Yasirkhan1301, 2020-10-17:
Nothing work for me, an error occured 

error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment
     mosiport    = portOutputRegister(digitalPinToPort(_din));
                 ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Yasirkhan1301, 2020-10-17:
Error gone by type casting every line
clkport = (unsigned char*)portOutputRegister(digitalPinToPort(_sclk));
mosiport = (unsigned char*)portOutputRegister(digitalPinToPort(_din));

ladyada, 2021-05-29:
this should be fixed now with the move to busio!

closed 2021-05-29 by ladyada