adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #29: Print

gittiago12, 2015-11-07:
How to print messages in display? My compiler is visual studio + visual micro plugin! This is the warnings:
Adafruit_PCD8544.h:69:46: error: expected class-name before '{' token
 class Adafruit_PCD8544 *:public Adafruit_GFX {
Sweet.ino:In function 'void setup()
Sweet.ino:11:10: error: 'class Adafruit_PCD8544' has no member named 'print
:display.print("Hello")
Error compiling
My project is called Sweet! Has no print or println method?

bbx10, 2015-11-07:
I have never used visual micro so you should ask over at esp8266.com or visual micro support forum.
This works with the Arduino IDE 1.6.5 or newer.

gittiago12, 2015-11-08:
yes the visual studio and visual micro work with any version of arduino ide! I made: clean the project, rebuild the project, clean project, restart pc and visual studio! now works fine!!!

micooke, 2016-02-19:
@gittiago12 - i use the same setup as you, never had an issue.
I suggest that the compiler is pointing to the wrong location, and that the error is in your code.
Try one of the examples.

micooke, 2017-09-07:
Someone close this issue please

closed 2019-07-22 by drak7