adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #32: PCBD8544 Nokia 5110 library not working in SAMD21 architecture (Arduino Zero)

qbiq, 2015-12-02:
Hello,

I have a Nokia 5110 Display, and I am trying to compile the example code (pcdtest.ino) in an Arduino Zero board, I assume that the library has not yet been ported to this new architecture.

Actually I have one of the first Arduino.org M0-Pro board, but both, the Arduino.org and Arduino.cc are pretty similar, as you already know.

I get the following complation error using Arduino IDE 1.6.5:
(Sorry that error messages are in Spanish, but I think the messages are easy to understand)

```javascript
Atención: platform.txt del núcleo 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contiene recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}" desfasada, automáticamente convertida a recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Considera el actualizar este núcleo.
C:\Users\quino\Documents\Arduino\libraries\arduino_183747\Adafruit_PCD8544.cpp: In member function 'void Adafruit_PCD8544::begin(uint8_t, uint8_t)':

C:\Users\qbiq\Documents\Arduino\libraries\arduino_183747\Adafruit_PCD8544.cpp:187:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     clkport     = portOutputRegister(digitalPinToPort(_sclk));

                 ^

C:\Users\qbiq\Documents\Arduino\libraries\arduino_183747\Adafruit_PCD8544.cpp:189:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'PortReg* {aka volatile unsigned char*}' in assignment

     mosiport    = portOutputRegister(digitalPinToPort(_din));

                 ^

exit status 1
Error de compilación
```

bbx10, 2015-12-03:
You can try my fixed version.

https://github.com/bbx10/Adafruit-PCD8544-Nokia-5110-LCD-library/tree/zero

qbiq, 2015-12-04:
Thank you, your version compiles my sketch!, everything solved!

erhanalankus, 2017-01-28:
This solution also works for MKR1000.

javiermelendrez, 2018-09-28:
I am trying to use your solution en esp8266 but it is not working!!

magiva, 2018-11-09:
this doesnt work for an esp32

PvdBerg1998, 2019-07-15:
Patching `Adafruit_PCD8544.h` lines 34 and 35 `uint8_t` to `uint32_t` should make it work.

drak7, 2019-07-29:
Fixed by #47.

closed 2019-07-29 by drak7