adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #36: Display only 13 characters instead of 14 on 1 line

pierre83, 2016-03-06:
With a PCD8544 ( 84 x 48) compliant device.
For example : display.print(F("12345678901234"));
Display : 1234567890123
Display : 4

bbx10, 2016-03-12:
This issue is caused by using Adafruit GFX library version1.1.x. Solution 1: Downgrade to version 1.0.2. Solution 2: Turn off text wrap then change print to println where needed. display.setTextWrap(false);
display.printfln("12345678901234");

I ended up using solution 2.

pierre83, 2016-03-12:
Thanks !

Good weekend !
  ----- Original Message ----- 
  From: bbx10 
  To: adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library 
  Cc: pierre83 
  Sent: Saturday, March 12, 2016 9:39 PM
  Subject: Re: [Adafruit-PCD8544-Nokia-5110-LCD-library] Display only 13 characters instead of 14 on 1 line (#36)

  This issue is caused by using Adafruit GFX library version1.1.x. Solution 1: Downgrade to version 1.0.2. Solution 2: Turn off text wrap then change print to println where needed. display.setTextWrap(false);
  display.printfln("12345678901234");

  I ended up using solution 2.

closed 2019-07-22 by drak7