adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #41: Issue using library with Arduino 101

paulhage, 2017-07-13:
I currently am using the Nokia LCD screen to display RFID tag information using an Arduino Uno. For some reason, when trying to use the same code with an Arduino 101, I face the following error messages:
Arduino: 1.8.2 (Mac OS X), Board: "Arduino/Genuino Uno"

Build options changed, rebuilding all
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino: In function ‘void loop()’:
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino:100:16: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         str[1] = 0x4400;
                ^
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino:93:10: warning: unused variable ‘i’ [-Wunused-variable]
    uchar i, tmp, checksum1;
          ^
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino:93:13: warning: unused variable ‘tmp’ [-Wunused-variable]
    uchar i, tmp, checksum1;
             ^
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino:96:15: warning: unused variable ‘RC_size’ [-Wunused-variable]
         uchar RC_size;
               ^
/Users/paulhage/Desktop/Addicore_RFID_Example/Addicore_RFID_Example.ino:97:15: warning: unused variable ‘blockAddr’ [-Wunused-variable]
         uchar blockAddr; //Selection operation block address 0 to 63
               ^
In file included from /Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/WString.h:34:0,
                 from /Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/Arduino.h:108,
                 from /Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.h:5,
                 from /Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:34:
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp: In member function ‘void Adafruit_GFX::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t, uint8_t)’:
/Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/avr/pgmspace.h:100:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
                                                            ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:60:42: note: in expansion of macro ‘pgm_read_dword’
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:753:43: note: in expansion of macro ‘pgm_read_pointer’
         GFXglyph *glyph  = &(((GFXglyph *)pgm_read_pointer(&gfxFont->glyph))[c]);
                                           ^
/Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/avr/pgmspace.h:100:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
                                                            ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:60:42: note: in expansion of macro ‘pgm_read_dword’
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:754:39: note: in expansion of macro ‘pgm_read_pointer’
         uint8_t  *bitmap = (uint8_t *)pgm_read_pointer(&gfxFont->bitmap);
                                       ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp: In member function ‘virtual size_t Adafruit_GFX::write(uint8_t)’:
/Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/avr/pgmspace.h:100:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
                                                            ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:60:42: note: in expansion of macro ‘pgm_read_dword’
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:837:50: note: in expansion of macro ‘pgm_read_pointer’
                 GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
                                                  ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp: In member function ‘void Adafruit_GFX::charBounds(char, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*)’:
/Users/paulhage/Library/Arduino15/packages/Intel/hardware/arc32/2.0.2/cores/arduino/avr/pgmspace.h:100:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long *)(addr))
                                                            ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:60:42: note: in expansion of macro ‘pgm_read_dword’
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-GFX-Library-master/Adafruit_GFX.cpp:952:50: note: in expansion of macro ‘pgm_read_pointer’
                 GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
                                                  ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-PCD8544-Nokia-5110-LCD-library-master/Adafruit_PCD8544.cpp: In member function ‘void Adafruit_PCD8544::begin(uint8_t, uint8_t)’:
/Users/paulhage/Documents/Arduino/libraries/Adafruit-PCD8544-Nokia-5110-LCD-library-master/Adafruit_PCD8544.cpp:192:17: error: cannot convert ‘uint32_t* {aka long unsigned int*}’ to ‘PortReg* {aka volatile unsigned char*}’ in assignment
     clkport     = portOutputRegister(digitalPinToPort(_sclk));
                 ^
/Users/paulhage/Documents/Arduino/libraries/Adafruit-PCD8544-Nokia-5110-LCD-library-master/Adafruit_PCD8544.cpp:194:17: error: cannot convert ‘uint32_t* {aka long unsigned int*}’ to ‘PortReg* {aka volatile unsigned char*}’ in assignment
     mosiport    = portOutputRegister(digitalPinToPort(_din));
                 ^
exit status 1
Error compiling for board Arduino/Genuino 101.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Is anyone aware of what may be causing this ? 
Does the library not work with the 101?

Any information would be appreciated.

njordan77, 2018-01-21:
having the same issue...tried many things...like go back to ESP8266 Core 2.3, did not improve.

drak7, 2019-07-29:
Should be fixed by #47, please open a new issue if you run into any problems.

closed 2019-07-29 by drak7