adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #44: Vertical mode?

arpruss, 2017-11-04:
As I look at the PCD8544 datasheet and the library code, I am a bit puzzled. The code sends command 0x20 to the PCD8544, which according to the [datasheet](http://eia.udg.edu/~forest/PCD8544_1.pdf) should put the PCD8544 in horizontal data mode (vertical data mode should be 0x22 per datasheet). But Adafruit_PCD8544::display() is sending data formatted as per the vertical mode specs. And yet it works fine on the cheap "Nokia 5110" display I bought an aliexpress. Still, it bothers me not knowing what is going on.

I wonder if the cheap Nokia 5110 displays don't use some clone controller that doesn't have a horizontal mode. I tried using 0x22 or 0x20, and it made no difference. But I still think it would be better if the code followed the datasheet and used the 0x22 command, even if 0x20 happens to work for the clones?

Or maybe I'm just being stupid and not understanding the code and/or datasheet.

mcauser, 2018-08-30:
You are correct in that sending `0x20` will put the display in horizontal addressing mode and `0x22` in vertical mode. `0x20` is the function set register and `0x02` is the bit that switches to vertical mode.

When switching mode, it does not affect what is currently on the display. It only affects where the next byte will be drawn.

The `pcd8544_buffer` in the library is storing the image in horizontal mode and `display()` loops over the 6 pages (each 8 rows tall) and writes up to 84 bytes (one per column) before moving to the next page. Each byte represents 8 vertical pixels. Partial updates is why they don't just set the x,y address to 0,0 and spi write the entire buffer.

I don't believe there are any cheap clones of the display. I've seen the same LCD wrapped in the same metal case on various coloured PCBs from various vendors. They have subtle differences in their backlight LEDs but are otherwise seemingly identical.

If you want to see the horizontal / vertical addressing in action, try setting the x,y position to 0,0 then spi writing a single byte `0xff`, pause and repeat. In both modes, each write will draw 8 vertical pixels. It's not possible to draw a row of 8 horizontal pixels from a single byte. They are always drawn vertical.

ladyada, 2021-05-29:
closing cause it seems to work?

closed 2021-05-29 by ladyada