adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #55: Teensy 3.0 and above support is needed

vamsee227, 2020-10-23:
Error compiling the test code for teensy board
Adafruit_PCD8544.cpp: In member function 'void Adafruit_PCD8544::begin(uint8_t, uint8_t)':
Adafruit_PCD8544.cpp:287:13: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'PortReg* {aka volatile long unsigned int*}' in assignment clkport = portOutputRegister(digitalPinToPort(_sclk));
Adafruit_PCD8544.cpp:289:14: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'PortReg* {aka volatile long unsigned int*}' in assignment mosiport = portOutputRegister(digitalPinToPort(_din));
Error compiling for board Teensy 3.2 / 3.1.

- Teensy board used:  **Teensy 3.2**

- Arduino IDE version **Arduino IDE ver. 1.8.13

vamsee227, 2020-10-23:
It worked after editing the header file (H file). After changing "uint32_t" in these lines
#else
typedef volatile uint32_t PortReg; ///< PortReg for other chips
typedef uint32_t PortMask;         ///< PortMask for other chips
#endif

Changed to "uint8_t"
#else
typedef volatile uint8_t PortReg; ///< PortReg for other chips
typedef uint8_t PortMask;         ///< PortMask for other chips
#endif

I request you to update the library by providing the support for Teensy 3.0 and above boards

Thanks 
Vamsee

drak7, 2020-10-23:
By default the example uses software SPI, switching to hardware SPI should fix it.

The port registers on the Teensy 3 are emulated as 8-bit for compatibility with Arduino so we could check for that board and use the 8-bit defines. The actual port registers for the teensy are 32-bit and can be used directly, that would be the better way to handle it. See [this teensy forum post](https://forum.pjrc.com/threads/17532-Tutorial-on-digital-I-O-ATMega-PIN-PORT-DDR-D-B-registers-vs-ARM-GPIO_PDIR-_PDOR?p=21228&viewfull=1#post21228) and [this one](https://forum.pjrc.com/printthread.php?t=48354&pp=25&page=1) for more info.

vamsee227, 2020-10-24:
Okay, thanks.

FerGT50, 2021-04-19:
Same problem here. Applying @vamsee227 corrections worked!
An update to the library to automatically account for 32 bit Teensies would be nice indeed!

Like:

`

#if defined(__SAM3X8E__) || defined(ARDUINO_ARCH_SAMD)
typedef volatile RwReg PortReg; ///< PortReg for SAMD
typedef uint32_t PortMask;      ///< PortMask for SAMD
#elif defined(__AVR__) || defined (TEENSYDUINO)
typedef volatile uint8_t PortReg; ///< PortReg for AVR
typedef uint8_t PortMask;         ///< PortMask for AVR
#else
typedef volatile uint32_t PortReg; ///< PortReg for other chips
typedef uint32_t PortMask;         ///< PortMask for other chips
#endif
`

ladyada, 2021-05-29:
this should be fixed now with the move to busio!

closed 2021-05-29 by ladyada