adafruit/Adafruit-PCD8544-Nokia-5110-LCD-library #60: Garbage on screen when drawing circles that are partially outside display

marcin-chwedczuk, 2021-06-12:
[image]

I was getting strange artifacts while running the library demo on Arduino Uno. Almost all images from demo worked but there was a glitch before the black circle demo. I looked at it more closely and it turned out to be caused by the `drawCircle` method.

Replacing the default demo with this code should illustrate the problem. Looks like an issue with proper cropping the pixels before sending them to the display:
```cpp
void testdrawcircle(void) {
  display.drawCircle(display.width()/2, display.height()/2, 40, BLACK);
  display.display();
  /*
  for (int16_t i=0; i<display.height(); i+=2) {
    Serial.print("i = "); Serial.println(i);
    display.drawCircle(display.width()/2, display.height()/2, i, BLACK);
    display.display();
  }
  */
  delay(3000);
}
```

I use some cheap PCD8544 clone from aliexpress without any markings. I run it on 3.3V with level shifters, the voltage and all looks fine.

ladyada, 2021-06-12:
hmm can you try previous versions of this library? we just did a major refactor!

marcin-chwedczuk, 2021-06-13:
Yeah only version 2.0.0 seems to be affected. Version 1.2.1 works perfectly.

kai-morich, 2021-12-19:
I observed similar issue with garbage when drawing partly off screen. This worked with 1.2.1, but creates garbage with 2.0.0

      display.setCursor(0, -1);
      display.println("text");

kai-morich, 2021-12-20:
looks like there is another issue for rotated display. I will try to investigate

kai-morich, 2021-12-20:
issue is solved

ladyada, 2021-12-20:
tahnks 😁

closed 2021-12-20 by ladyada